ARMmbed/mbed-os-5-docs #711: nitpicks in documentation of device management features in Mbed CLI

maclobdell, 2018-09-06:
On the page: 
https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/tools/offline/cli-device-management.md

Nitpick 1 - 
"Arm Mbed OS allows you to update your device firmware, enabled by our Pelion IoT platform."

The use of "our" goes against the ecosystem viewpoint that Mbed OS is community/partnership driven project.  Saying "our" implies that Arm has full control over both Mbed OS and Pelion, which for Mbed OS is not true.  I recommend dropping the "our".  Yes, this is a nitpick. 

Nitpick 2 -
```
$ mbed config -G CLOUD_SDK_API_KEY <API_KEY>
$ mbed target K64F
$ mbed toolchain GCC_ARM
```
Can we please stop using K64F in our general documentation?  I love the K64F, but general instructions should be generalized to apply to any platform.

I recommend changing this to: 
```
$ mbed config -G CLOUD_SDK_API_KEY <API_KEY>
$ mbed target <TARGET>
$ mbed toolchain <TOOLCHAIN>
```

AnotherButler, 2018-09-06:
@maclobdell The phrasing in the first part comes directly from our product team, so I'll confirm with them that it's OK to change. Your second point is spot on. 👍 Good catch.

AnotherButler, 2018-09-11:
Thanks @maclobdell 
This is resolved by PR #732 
Please feel free to reopen if this PR does not satisfy these requirements.

closed 2018-09-11 by AnotherButler