ARMmbed/mbed-os-5-docs #718: Enhancements on how to use unittest

MarceloSalazar, 2018-09-10:
I'm reviewing docs as of 5.10 OOB.

The docs for unittest should be tweaked to explain how to use it.

- Missing instructions on how to run unittest with Mbed CLI, to be coherent with greentea and icetea docs.
https://os-doc-builder.test.mbed.com/docs/development/tools/unit-testing.html

- Mbed CLI is required. Not correct statement here:
https://github.com/ARMmbed/mbed-os/blob/master/UNITTESTS/README.md

- The README.md should be shown in os.mbed.com/docs , this is very good information.

- Introduction to testing (piramid diagram) is good but should be moved one level up in the docs as general intro.

@OPpuolitaival @theotherjimmy  @AnotherButler

lorjala, 2018-09-10:
@MarceloSalazar I will update the documentation.

> Missing instructions on how to run unittest with Mbed CLI, to be coherent with greentea and icetea docs.

Yes, I will add instructions.

> Mbed CLI is required.

Not required, only encouraged. The Mbed CLI command just works as an alias for a python script. All following use cases are supported:

- Mbed CLI command: mbed test --unittests
- Python script in Mbed OS repository: UNITTESTS/mbed_unittest.py
- Manually build and run tests with CMake and make tools.

> The README.md should be shown in os.mbed.com/docs , this is very good information.

I will mirror the documentation, so both GitHub and the docs will contain the same information.

> Introduction to testing (piramid diagram) is good but should be moved one level up in the docs as general intro.

The diagram will move from unit testing documentation to testing intro.

MarceloSalazar, 2018-09-10:
@lorjala. Thanks for the comments.

> I will mirror the documentation, so both GitHub and the docs will contain the same information.

Instead of duplicating the doc, do you think could be added to the docs' json index to pull-in docs automatically?

lorjala, 2018-09-10:
@MarceloSalazar Yes, it could be added docs.json. I will start merging and editing the docs.

Also, why does docs for greentea and icetea exist separately when they are also mirrors from GitHub documentations?

AnotherButler, 2018-09-10:
@lorjala @MarceloSalazar We have instructions about using unit tests with Mbed CLI: https://os-doc-builder.test.mbed.com/docs/development/tools/test-and-debug.html
Please link to this instead of duplicating content.

@MarceloSalazar Great suggestion moving the pyramid diagram up a level. After looking over it, I agree that's the best place for this content.

I'm not sure I agree with pulling in the README. This would be a large last-minute addition, and I don't know if it's what's best for our docs as it doesn't fit our structure and includes duplicate and not-future-proof information.

AnotherButler, 2018-09-11:
PR #734 resolves the diagram moving. PR #735 adds the link to the instructions about using unit tests with Mbed CLI.

AnotherButler, 2018-09-11:
Please feel free to reopen if these PRs do not resolve this issue.

closed 2018-09-11 by AnotherButler