ARMmbed/mbed-os-5-docs #720: Mbed CLI device management - missing pointers Python Cloud SDK

MarceloSalazar, 2018-09-10:
I'm evaluating "Updating devices with Arm Mbed CLI" during 5.10 OOB and reviewing this doc:
https://os-doc-builder.test.mbed.com/docs/development/tools/cli-update.html

It needs to have pointers on how to setup Python Cloud SDK. 
https://github.com/ARMmbed/mbed-cloud-sdk-python

theotherjimmy, 2018-09-10:
@MarceloSalazar The setup, adding an api key, is already part of the documentation. Why mention that we're using the python SDK?

theotherjimmy, 2018-09-11:
Oh, we're talking about installing it. It should be in `requirements.txt` so a usual `pip -Ur requirements.txt` should work.

MarceloSalazar, 2018-09-11:
Yes, I believe should be installed as part of `requirements.txt`, this doesn't seem to work automatically and requires the user invoking `pip ...`, right?

theotherjimmy, 2018-09-11:
"does not seem to" implies that you have tested it and found that this is not the case. If the auto-install fails, then a user will have to run pip, yes.

AnotherButler, 2018-09-11:
Does this PR, which adds a link to the Pelion quick start, resolve this issue?
https://github.com/ARMmbed/mbed-os-5-docs/pull/733

AnotherButler, 2018-09-12:
ping @MarceloSalazar @theotherjimmy

MarceloSalazar, 2018-09-12:
I don't think so, that PR is about the SDK API key.
And I was referring to the Mbed Cloud SDK Python **package**

However, as @theotherjimmy  commented, it should be installed automatically, or the user may need to install it with pip.

I'm happy to close this item with nor further action.

closed 2018-09-12 by AnotherButler