ARMmbed/mbed-os-5-docs #725: Review of Updating devices with Arm Mbed CLI section in 5.10 oob mbed-os docs

vijhar01, 2018-09-11:
1. https://os-doc-builder.test.mbed.com/docs/development/tools/cli-update.html (Project Setup section). Regarding " Obtain the API key from the the Device Management Portal"

Consider including steps/screenshots on how to get the API key from portal. 

2. "Device Management Portal" portal link in the page -  https://os-doc-builder.test.mbed.com/docs/development/tools/cli-update.html takes you to https://cloud.mbed.com/docs/current/introduction/index.html; Should it be linking https://portal.mbedcloud.com/login?

AnotherButler, 2018-09-11:
@vijhar01 Good points. 👍 

Do the links in this PR make more sense?
https://github.com/ARMmbed/mbed-os-5-docs/pull/733

vijhar01, 2018-09-12:
@AnotherButler 
Yes. It does. Thanks. I will close this now.

closed 2018-09-12 by vijhar01