ARMmbed/mbed-os-5-docs #749: TCPSocket class hierarchy in 5.9 is incorrect

toyowata, 2018-09-27:
API document in 5.9 show incorrect class hierarchy for the TCPSocket class.  It is inherited from the InternetSocket class, but it is inherited from the Socket class in actual code.

API document (5.9)
https://os.mbed.com/docs/v5.9/reference/tcpsocket.html
Code (5.9.x)
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.9/features/netsocket/TCPSocket.h#L31

API document in the 5.10 is correct.

API document (5.10)
https://os.mbed.com/docs/v5.10/apis/tcpsocket.html
Code (5.10.x)
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.10/features/netsocket/TCPSocket.h#L31

kegilbert, 2018-09-27:
Looked at this a bit with @AnotherButler and passed it off to @cmonr for more info, but looks like that change was made in master but was not pulled into the 5.9 release branch. I'm not 100% clear on where the docs pull from during release doc builds (master versus RC branch). Assuming this change in particular did not go into 5.9 due to code freeze timing (was merged ~3 weeks before the release went out). Cruz can comment more on that likely as well.

cmonr, 2018-09-27:
Will see if I can't pin this down tomorrow.

I'm not too familiar with how the docs are generated, but if the doxygen was pointing towards the master branch after this (https://github.com/ARMmbed/mbed-os/pull/7192) was merged (note, merged != released), then I could see where the issue came up.

We already confirmed that the mbed-os-5.9 branch's API didn't change throughout it's lifecycle, which is why I suspect the above occurred.

AnotherButler, 2018-09-28:
After discussing this with @iriark01 , we've decided to update the docs .json to point to the release branch instead of master. This should prevent this issue in the future.

AnotherButler, 2018-09-28:
@toyowata Thank you for raising this issue. We have implemented this fix, and now the documentation reflects the code: https://os.mbed.com/docs/v5.9/reference/tcpsocket.html.

closed 2018-09-28 by AnotherButler