ARMmbed/mbed-os-5-docs #816: Broken link for Mbed CLI installer

toyowata, 2018-11-06:
An URL for the Mbed CLI installer (Windows) is broken here:
https://os.mbed.com/docs/v5.10/tools/installation-and-setup.html

When I clicked `Windows installer for Mbed CLI`, I got following message:
```
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>NoSuchKey</Code>
<Message>The specified key does not exist.</Message>
<Key>builds/Mbed_installer_v0.4.7.exe</Key>
<RequestId>E7C1DF1CD3F06C14</RequestId>
<HostId>
hxhwmMzploXwO9i4cFbBJEnVMq51PasW3Ne4M2Z/S+SFY4B3HRfHFruD1cpco7y5Yf9lK2/JDOo=
</HostId>
</Error>
```

Source is:
https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/tools/CLI/cli-setup.md/cli-setup.md

And this also has broken link:
https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/tools/CLI/cli-setup.md/cli-reqs.md

AnotherButler, 2018-11-06:
@theotherjimmy is addressing this.

theotherjimmy, 2018-11-06:
Use https://github.com/ARMmbed/mbed-cli-windows-installer/releases/latest

AnotherButler, 2018-11-06:
Closing because this issue is now resolved

closed 2018-11-06 by AnotherButler