ARMmbed/mbed-os-5-docs #884: Code guidelines - enhancement

0xc0170, 2018-12-20:
# Docs issue template

Request comes from https://github.com/ARMmbed/mbed-os/pull/9158 

## Page link

## Expected to find...

> It might be useful to extend the code guidelines to mention that data structures local to a C++ files should be declared inside an anonymous namespace

## Found

No mention of anonymous namespace

cc @kjbracey-arm @pan- @SenRamakri

bulislaw, 2019-01-03:
@kjbracey-arm @pan- @deepikabhavnani I remember you being involved in some changes to namespaces recently, does anyone want to have a stab at this?

AnotherButler, 2019-03-27:
@0xc0170 I'm closing this because the PR addressing it has merged. Please reopen if it has not been fully resolved.

closed 2019-03-27 by AnotherButler