ARMmbed/mbed-os-5-docs #924: Question: Sources for Mbed_OS_diagram_for_intro.png

loverdeg-ep, 2019-01-16:
# Docs issue template
@AnotherButler @BlackstoneEngineering 

Any chance of the following diagram sources being released with this repository? Hoping they are Draw.io.

[image]

Nice thing about draw.IO is that the diagrams can be exported as .svg, still be relatively human readable, as well as rendered by github/gitlab, and imported back into draw.io. Single source of truth.

I've done similar diagrams and can appreciate the effort that went into it.

I'd like to use bits and pieces of it underneath and above some of our own software/hardware diagrams.
Whatever license this is released under would of course be respected.

## Page link
https://s3-us-west-2.amazonaws.com/mbed-os-docs-images/Mbed_OS_diagram_for_intro.png

loverdeg-ep, 2019-01-18:
@AnotherButler happening?

AnotherButler, 2019-01-22:
Unfortunately, this image was not created in draw.io or anything that's easy to take apart. If you'd like to submit a proposal with how you'd like to use the content, there's a chance we could create something for you.

loverdeg-ep, 2019-01-22:
Ahh, MS Paint...

Ehh... I'll probably just re-create and maybe do an MR if I ever get around to it.
We'd be using it probably exclusively for internal design documents, but I suppose it could leak into development guides.

AnotherButler, 2019-02-08:
@loverdeg-ep Because this isn't in draw.io, shall I close this issue?

closed 2019-02-08 by loverdeg-ep