ARMmbed/mbed-os-5-docs #939: List available toolchains for exporting via mbed cli

ASealion, 2019-02-06:
# the command mbed export -h did not list the supported toolchains

The documentation says
> For a complete list of supported export toolchains, you can run:
but provided command will only give an overview of all options available

If you use an invalid toolchain as -i parameter you get the list of valid toolchains

## https://os.mbed.com/docs/mbed-os/v5.11/tools/exporting.html

## List of supported toolchains

## Only general help on mbed export

AnotherButler, 2019-02-08:
@theotherjimmy or @bridadan Could one of y'all please help with this?

bridadan, 2019-02-11:
Hi @ASealion,

I agree, the language is confusing, `mbed export -h` only lists the available options, not all of the available IDEs.

Turns out we can run `mbed export --supported ides` on most recent releases of Mbed OS (since 5.5.0):

```
mbed export --supported ides
[mbed] Working path "C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-cloud-client-example" (program)
cces
cmake_gcc_arm
codeblocks
e2studio
eclipse_armc5
eclipse_gcc_arm
eclipse_iar
embitz
gnuarmeclipse
iar
make_armc5
make_armc6
make_gcc_arm
make_iar
mcuxpresso
netbeans
qtcreator
sw4stm32
uvision5
uvision6
vscode_armc5
vscode_gcc_arm
vscode_iar
```

Would that be a viable substitute? If so, this would be the line to change: https://github.com/ARMmbed/mbed-os-5-docs/blame/f12ac64625e2770a848d1b819ceee29b9b72c8a6/docs/tools/toolchains/export_to_third_party.md#L57

AnotherButler, 2019-02-21:
@ASealion Does this substitute work for you?

ASealion, 2019-02-22:
@AnotherButler Yea, that's the command / "change" I was looking for.

AnotherButler, 2019-02-27:
@bridadan @theotherjimmy Could one of you please make this change to the docs?

theotherjimmy, 2019-02-27:
Sure. Just a minute.

theotherjimmy, 2019-02-27:
@AnotherButler see https://github.com/ARMmbed/mbed-os-5-docs/pull/970

AnotherButler, 2019-02-27:
Resolved with PR #970 
You can see the updated document at the link in the description.

closed 2019-02-27 by AnotherButler