ARMmbed/mbed-os-5-docs #942: Add this repository to ARMmbed pinned repo's

loverdeg-ep, 2019-02-06:
# Docs issue template

## Page link

https://github.com/ARMmbed

## Would like to find...

mbed-os-5-docs pinned at top. I've got a bookmark but...

## Found

[image]

AnotherButler, 2019-02-06:
I think this is a great suggestion. @ARMmbed/mbed-os-maintainers What are your thoughts? Is this something we can do?

closed 2019-02-07 by ciarmcom

loverdeg-ep, 2019-08-07:
@adbridge Can this be adjusted?
@ARMmbed/mbed-os-maintainers 

People probably look for the docs more than they do minimal-printf

[image]

evedon, 2019-08-07:
It's my bad I think! 
I changed that recently thinking these repos were pinned for me only... ! 
I will revert to the ones shown at the top, plus the mbed-os-5-docs

loverdeg-ep, 2019-08-07:
Hahha, NP. I had a feeling somebody changed them thinking they were per user.