ARMmbed/mbed-os-5-docs #991: custom_targets and configuration intent for custom hardware

kiranshila, 2019-03-06:
## Page link
https://os.mbed.com/docs/mbed-os/v5.11/reference/adding-and-configuring-targets.html
https://os.mbed.com/docs/mbed-os/v5.11/evaluation-production/stage-3-prototype.html

## Expected to find...
A more thoughtful explanation as to why one would need to configure a custom target rather than a brief on JSON and inheritance.

I can not seem to find a straightforward explanation on how to build an mbed project with the CLI for a custom PCB that uses an MCU similar to an existing target. I assume as all of the STM targets are dev boards that the approach that is consistent with the mbed design philosophy is to create a custom target that slightly modifies the dev board target. If this is the case, then surely there needs to be some explanation as to this fact.

This is different than the idea enumerated in https://os.mbed.com/docs/mbed-os/v5.11/evaluation-production/stage-3-prototype.html as there is really another option besides writing an entire new target. Especially if the targets that are meant to be in mbed-os/targets are not generic MCU targets but rather development platforms, should it really be encouraged to add a target to custom hardware inside of a git tracked directory? That doesn't seem anywhere near as portable as defining a custom target with the json file in the project directory.

AnotherButler, 2019-03-06:
@kiranshila Thanks for raising this issue. I agree that this is content we should add. I'll raise this with my team. In the meantime, we do have a [porting guide about adding custom targets from scratch](https://os.mbed.com/docs/mbed-os/v5.11/porting/index.html), but I don't think it quite meets your needs.

dlfryar, 2019-03-07:
Hi @kiranshila,

An unofficial public example doing this on the Nordic nRF52840 is here:

https://github.com/dlfryar/mbed-os-example-max77801/tree/MM-custom_targets

The other option is to do the same in the mbed-os structure without the custom_targets.json config:

https://github.com/dlfryar/mbed-os/tree/target_mm/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_MM

If you have specific recommendations to make it more clear we happily accept pulls requests.  Feel free to fork and submit.

kiranshila, 2019-03-07:
Awesome, thanks!

I'm almost done defining a new custom target, once I finish I would love to write a more formal write up of the process in the context of moving from prototype to production.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

AnotherButler, 2019-12-19:
@kiranshila We believe our new guide to [porting a custom board](https://os.mbed.com/docs/mbed-os/v5.15/porting/porting-a-custom-board.html) resolves this issue. Please reopen it if it has not been resolved.

closed 2019-12-19 by AnotherButler