ARMmbed/mbed-os-5-docs #1005: Old article doesn't redirect correctly

carlosperate, 2019-03-11:
# Docs issue template

## Page link

Original URL: https://docs.mbed.com/docs/mbed-os-handbook/en/5.4/debugging/debugging_microbit/

## Expected to find...

Should redirect to: https://os.mbed.com/docs/mbed-os/v5.11/tutorials/debug-microbit.html

## Found

Instead it redirects to: https://os.mbed.com/docs/mbed-os/v5.11/introduction/index.html

AnotherButler, 2019-03-11:
@JoeTheGuitarist I don't think I can fix this one. Could someone on your team please look into this?

JoeTheGuitarist, 2019-03-11:
@AnotherButler yes, I'll add a ticket.

carlosperate, 2019-03-11:
Thanks @AnotherButler and @JoeTheGuitarist!

AnotherButler, 2019-08-16:
@carlosperate Thanks for raising this issue. We've implemented redirects for documentation for Mbed OS 5.6 and later. We have no plans at this time to implement redirects for documentation before Mbed OS 5.6.

closed 2019-08-16 by AnotherButler