ARMmbed/mbed-os-5-docs #1011: mbed-cli python3 requirement not supported with Ubuntu 16.04 LST

teetak01, 2019-03-14:
Docs say:

Python: Mbed CLI is a Python script, so you'll need Python to use it. We test Mbed CLI with Python versions 2.7.11, and with 3.6.5 and greater.

But Ubuntu 16.04 LTS does not support Python 3.6.5 out-of-box.

https://packages.ubuntu.com/xenial/python3
python3-minimal (= 3.5.1-3)

Released | End of Life
-- | --
Ubuntu 16.04 LTS | April 2016 | April 2021

teetak01, 2019-03-14:
cc: @OPpuolitaival

JanneKiiskila, 2019-03-14:
One question is - do we know ANY reason why we want Python 3.6.5 vs. Python 3.5.2? Is there some feature we want there or?

It seems to me that someone has sort of made their mind that Ubuntu 18.04 is the way to go?

0xc0170, 2019-03-14:
@ARMmbed/mbed-os-tools Please review

Looking at cli definitions - https://github.com/ARMmbed/mbed-cli/blob/3b43228d6f21d99a4919866c697c0f73cecee313/setup.py#L35 - few versions not supported but 3.5 should be. Circle CI there defines 3.6 and 3.7 testing.

JanneKiiskila, 2019-03-14:
There's an opportunity here at the same time - I would personally drop the Python2 support completely and go with 3 only. That would make at least the full client story a bit smoother, as the Factory tooling doesn't support Python2 any way.

theotherjimmy, 2019-03-14:
@JanneKiiskila I'm happy to drop python2, as soon as our testing infra is on python3 and the installers include python3 and we have a migration path for all releases prior to 5.11.x (when python 3 was truly beginning to be usable)

cmonr, 2019-03-14:
@JanneKiiskila We've started talking about when would be the right time to drop Py2 support, but unless there's a critical issue that it would fix, I don't see that happening for this release.

In Mbed OS, Travis CI tests the following versions: Py 2.7.14, 3.5.6, 3.6.3, and 3.7.1.
As @0xc0170 mentioned as well, Mbed CLI is testing Py3 as well.

> It seems to me that someone has sort of made their mind that Ubuntu 18.04 is the way to go?
I suspect it's probably more like noone specificed a base minimum version that should be supported.

theotherjimmy, 2019-03-18:
@teetak01 What action do you expect to this issue? Do you expect that line to include python 3.5.x? do you expect me to say, yes that's what we support?

teetak01, 2019-03-18:
Just some clarification in the documentation that which python3 versions we actually promise to support, and is Ubuntu 16.04 included in that range (which I think should be, even if there might not have been strict requirement for it).

This will impact what requirements we define on Cloud side, as I prefer to align those with the OS side.

theotherjimmy, 2019-03-18:
@teetak01 It seems to me that you're quoting the section of the documentation that documents the requirement. Is there somewhere else you want it documented?

teetak01, 2019-03-18:
As long as we have clear idea what we promise to support, and it can be found from the official documentation for Mbed CLI somewhere there.

I opened this mainly because it did not seem to be clear what we actually support.

theotherjimmy, 2019-03-18:
https://os.mbed.com/docs/mbed-os/v5.11/tools/linux.html This is the page you quoted with:
>We test Mbed CLI with Python versions 2.7.11, and with 3.6.5 and greater.

Is that sufficient?

teetak01, 2019-03-19:
@theotherjimmy 
> We test Mbed CLI with Python versions 2.7.11, and with **3.5.1** and greater.

Of course I cannot really say what it should be, that is what I am asking from OS team.

theotherjimmy, 2019-03-19:
@teetak01 In the future, please don't phrase requests like this as a question; you were not asking for clarity, you were already clear on what's supported and you wanted to change it. Instead just ask that we support 3.5.1 and greater instead. That way we don't have to talk about it a bunch to get to that conclusion.

We can probably drop that to 3.5.1, after we change the version used in CI.

iriark01, 2019-03-22:
This doesn't strike me as a docs issue. Can I close it?

theotherjimmy, 2019-03-22:
@iriark01 This has changed into an engineering issue to support Python 3.5.1. @teetak01 Is that correct?

closed 2019-03-22 by theotherjimmy