ARMmbed/mbed-os-5-docs #1035: Debugging page has a link to itself instead of to expanded page

mpg, 2019-03-27:
# Docs issue template

## Page link

https://os.mbed.com/docs/mbed-os/v5.8/tutorials/debugging.html

## Expected to find...

In the sentence "Read our tutorial, which guides you through this process." at the very end of the page, I expect the link "our tutorial" to take me to another page with more information the process.

## Found

The links "our tutorial" brings me back to the same page and I'm not getting the extra information that I was so eager to learn.

AnotherButler, 2019-03-27:
@mpg Thanks for raising this issue. 

That content continues in the rest of the pages in that section. If you're looking to specifically export the project to a third party IDE for debugging, please see the [Eclipse](https://os.mbed.com/docs/mbed-os/v5.8/tutorials/eclipse.html), [Keil uVision](https://os.mbed.com/docs/mbed-os/v5.8/tutorials/keil-uvision.html), [Visual Studio Code](https://os.mbed.com/docs/mbed-os/v5.8/tutorials/visual-studio-code.html) or [BBC micro:bit](https://os.mbed.com/docs/mbed-os/v5.8/tutorials/debug-microbit.html) page.

I see that you're looking at the 5.8 documentation. If you'd like the latest version of the documentation, instead, please replace "v5.8" with "latest" in the URL.

I've also removed the self-referencing link on all of the affected versions. If this does not resolve the issue, please feel free to reopen it with more information.

closed 2019-03-27 by AnotherButler

mpg, 2019-03-28:
@AnotherButler Thanks for fixing so quickly! And thanks for the tip about "latest" in the URL.