ARMmbed/mbed-os-5-docs #1038: mbed import could not import from https://os.mbed.com

turmary, 2019-04-05:
# Docs issue template

## Page link

## Expected to find...

## Found
tary@tary-think MINGW32 /f/home/tary/mbed
$ git clone https://os.mbed.com/users/hudakz/code/CAN_Hello
Cloning into 'CAN_Hello'...
fatal: unable to update url base from redirection:
  asked for: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs?service=git-upload-pack
   redirect: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs/?service=git-upload-pack

tary@tary-think MINGW32 /f/home/tary/mbed
$ git clone http://os.mbed.com/users/hudakz/code/CAN_Hello
Cloning into 'CAN_Hello'...
fatal: unable to update url base from redirection:
  asked for: http://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs?service=git-upload-pack
   redirect: https://os.mbed.com/users/hudakz/code/CAN_Hello/info/refs/?service=git-upload-pack

tary@tary-think MINGW32 /f/home/tary/mbed
$ mbed import http://os.mbed.com/users/hudakz/code/CAN_Hello
[mbed] Working path "F:\home\tary\mbed" (directory)
[mbed] Program path "F:\home\tary\mbed"
[mbed] Importing program "CAN_Hello" from "https://os.mbed.com/users/hudakz/code/CAN_Hello" at latest revision in the current branch
[mbed] ERROR: Unable to clone repository (http://os.mbed.com/users/hudakz/code/CAN_Hello)
---

tary@tary-think MINGW32 /f/home/tary/mbed
$ mbed import https://os.mbed.com/users/hudakz/code/CAN_Hello
[mbed] Working path "F:\home\tary\mbed" (directory)
[mbed] Program path "F:\home\tary\mbed"
[mbed] Importing program "CAN_Hello" from "https://os.mbed.com/users/hudakz/code/CAN_Hello" at latest revision in the current branch
[mbed] ERROR: Unable to clone repository (https://os.mbed.com/users/hudakz/code/CAN_Hello)
---

AnotherButler, 2019-04-05:
@SenRamakri Could someone from your team help with this?

cmonr, 2019-04-08:
@turmary Please reopen this issue in Mbed OS: https://github.com/ARMmbed/mbed-os/

This does not appear to be a tool issue, not doc issue.

closed 2019-04-08 by cmonr