ARMmbed/mbed-os-5-docs #1078: Connectivity configuration documentation is lost

SeppoTakalo, 2019-06-05:
If you compare the documentation regarding connectivity configuration, you will notice that between 5.11 and 5.12 the sections were completely lost and mixed up with unrelated stuff.

For comparison, in Mbed OS 5.11 this page: https://os.mbed.com/docs/mbed-os/v5.11/reference/configuration-connectivity.html
contains clear sections and internal table of contents.
You can easily link to relevant sections like this: [Selecting the default interface](https://os.mbed.com/docs/mbed-os/v5.11/reference/configuration-connectivity.html#selecting-the-default-network-interface)

When you compare that to 5.12 documentation: https://os.mbed.com/docs/mbed-os/v5.12/reference/configuration-connectivity.html
All the sections are lost, content is just one big blob that could be produced by running `mbed compile --config -v` and therefore the page does not offer any extra information.

Also, it contains unrelated parameters like `target.boot-stack-size` which should be in platform configuration page.

This page does not anymore document anything. It is just a dump from a script.

AnotherButler, 2019-06-05:
The code parameters are pulled straight from the code base. If this page contains incorrect parameters, it means they are incorrect in the code or the script is wrong. Could you please look at the script to see which is the case?

If you or one of your engineers can modify the script to add sections, that could be helpful.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

iriark01, 2020-04-20:
@SeppoTakalo unless you or someone else from engineering can modify the script to add sections, I'll close this.

closed 2020-05-27 by iriark01