ARMmbed/mbed-os-5-docs #1098: Is it really expected that we implement a deprecated function?

sjlongland, 2019-06-27:
Hi,

I was just reading https://os.mbed.com/docs/mbed-os/v5.12/porting/hardware-accelerated-crypto.html in regards to putting hardware acceleration in OpenThread which uses mbedTLS 2.13.0, specifically AES and SHA-256 algorithms (ultimately ECC too, soon as I figure out how).

The documentation states:
> Mbed TLS has separate modules for the different cryptographic primitives. Hardware acceleration interface is available for the following modules and functions:
>
>    Symmetric
>        AES: mbedtls_aes_setkey_enc(), mbedtls_aes_setkey_dec(), mbedtls_internal_aes_encrypt(), mbedtls_internal_aes_decrypt().
…
>        SHA256: mbedtls_sha256_process().

For SHA-256, that takes me [here](https://tls.mbed.org/api/sha256_8h.html#a12db056a85dc91162ebfbe41c550c1de) which states:

> Deprecated:
>    Superseded by mbedtls_internal_sha256_process() in 2.7.0.

Is there an updated hardware acceleration guide I should be following instead?

sjlongland, 2019-06-27:
Also whilst I'm here… the module I'm talking to (AES in CC2538) does *NOT* implement SHA-224… do I need to take care of the initial value when initialising the module or is the first block I read going to contain the initialisation data for SHA-224?

AnotherButler, 2019-06-28:
@Patater @RonEld Could one of you help with this?

RonEld, 2019-06-30:
Hi @sjlongland 
Thank you for your question!
The reason for hte deprecation of the function is for backwards compatibility of existing applications that haven't moved to the new API (`mbedtls_internal_sha256_process()`) . These deprecated functions are intended to be removed in a release or two, so I wouldn't bother in implementating these functions in your alternative implementation.
I would say the guidelines need to be updated, unfortunately.

>Also whilst I'm here… the module I'm talking to (AES in CC2538) does NOT implement SHA-224… do I need to take care of the initial value when initialising the module or is the first block I read going to contain the initialisation data for SHA-224?

You have three options:
1. Yes, take care of the initial value, depending on your platform's implementation and design.
2. Have your API return `MBEDTLS_ERR_PLATFORM_FEATURE_UNSUPPORTED` for sha224, in this case your driver will not support sha224, if that's ok with you.
3. Fallback to sw implementation for sha224. I wouldn't recomend this, as it increases memory usage, with a minimal added value.
You can find additional guidelines for hw accelerations, targeting the standalone Mbed TLS [here](https://tls.mbed.org/kb/development/hw_acc_guidelines). Porting an alternative implementation for the standalone Mbed TLS is mostly identical to the way you do it in Mbed OS, except where you define the `*_ALT` and location of the implementations.

sjlongland, 2019-07-01:
> Thank you for your question!
> The reason for hte deprecation of the function is for backwards compatibility of existing applications > that haven't moved to the new API [...]

Okay, so should I implement `mbedtls_internal_sha256_process` instead?

> > Also whilst I'm here… the module I'm talking to (AES in CC2538) does NOT implement SHA-224… do I need to take care of the initial value when initialising the module or is the first block I read going to contain the initialisation data for SHA-224?
> Yes, take care of the initial value, depending on your platform's implementation and design.

Makes sense. :-)  I'm not sure that SHA-224 is being used in OpenThread, but it's probably better to handle it anyway.

RonEld, 2019-07-01:
>Okay, so should I implement `mbedtls_internal_sha256_process()` instead?

This is an internal function, but you will need it to build. You should have it return `MBEDTLS_ERR_PLATFORM_FEATURE_UNSUPPORTED`. See [this port](https://github.com/ARMmbed/mbed-os/blob/master/features/cryptocell/FEATURE_CRYPTOCELL310/sha256_alt.c#L57) for reference.

>Makes sense. :-) I'm not sure that SHA-224 is being used in OpenThread, but it's probably better to handle it anyway.

If your use case is OpneThread only, and it doesn't require SHA-224, I wouldn't bother in setting the initial value, and just return `MBEDTLS_ERR_PLATFORM_FEATURE_UNSUPPORTED` for SHA-224. If your hw also sets initial value within, you might get into needless issues on your porting effort.  If SHA-224 is needed, then you should do these efforts, I guess

sjlongland, 2019-07-01:
No problems, that looks like a good starting point.  Many thanks. :-)

AnotherButler, 2019-07-18:
We have updated content through PR #1115

AnotherButler, 2019-07-19:
Please feel free to reopen this if you feel it has not been resolved.

closed 2019-07-19 by AnotherButler