ARMmbed/mbed-os-5-docs #1100: Please add UARTSerial to the docs and clarify serial handling

frmdstryr, 2019-07-03:
I wasted half a day trying to get `RawSerial` to work with interrupts only to find that it does not work reliably. After digging through various issues I finally found that `UARTSerial` with sigio should be used instead. `UARTSerial` is not mentioned anywhere in any of the "Serial" related doc pages. 

Also it's very confusing what the difference is between `RawSerial`, `Serial`, `USBSerial`, and `UARTSerial`.  An explanation should of the differences between these and when to use each be somewhere (and if it is somewhere it's not easy to find). 

Also it's unclear what happens when a `serial(USBTX, USBRX)` is defined in user code (as is done in all the Serial examples). Does this override the stdin/stdout serial, conflict with it, etc..?

## Page link

https://os.mbed.com/docs/mbed-os/v5.13/apis/index.html

## Expected to find...

At least _some_ reference to UARTSerial as mentioned in https://os.mbed.com/docs/mbed-os/v5.13/apis/filehandle.html

## Found

Nothing.

Also https://github.com/ARMmbed/mbed-os-5-docs/pull/336 which is closed, why?

0xc0170, 2019-07-04:
Thanks for reporting, we will review and fix this (I've noticed few other similar reports recently in our issue tracker).

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

Masitech, 2020-02-17:
am trying to get UARTserial to work with custom nrf52832 but having no success. I got serial and rawserial to work however they are not buffered.

AnotherButler, 2020-03-09:
@Masitech Thanks for commenting 👍 We now have documentation for BufferedSerial in our v6.0-preview branch: https://os.mbed.com/docs/mbed-os/v6.0-preview/apis/bufferedserial.html

Please note this branch is a preview and a work in progress.

AnotherButler, 2020-03-16:
We believe we answered this question, so I'm closing this issue. Please feel free to reopen it if it has not been resolved.

closed 2020-03-16 by AnotherButler