ARMmbed/mbed-os-5-docs #1101: SDBlockDevice example doesn't import

pabloassembler, 2019-07-04:
SDBlockDevice example on

https://os.mbed.com/docs/mbed-os/v5.13/apis/sdblockdevice.html

doesn't import as a program. Instead, a full directory os examples is imported.

AnotherButler, 2019-07-09:
@pabloassembler Thank you for raising this issue. We're looking into it.

AnotherButler, 2019-07-11:
Thank you for bringing this to our attention. For this example, we have disabled the button causing the unexpected behavior and ask that you [copy the example into a new program](https://os.mbed.com/docs/mbed-os/v5.13/tools/creating-a-new-program.html) rather than importing it directly.

AnotherButler, 2019-07-15:
Please feel free to reopen this issue if you feel it has not been resolved.

closed 2019-07-15 by AnotherButler