ARMmbed/mbed-os-5-docs #1129: Add sudo to install mbed-cli

doom-fr, 2019-08-07:
# Docs issue template
command `pip install mbed-cli` does not work on `Ubuntu 18.04.LTS`

## Page link
https://os.mbed.com/docs/mbed-os/v5.11/tools/linux.html

## Expected to find...
`sudo pip install mbed-cli`

## Found
`pip install mbed-cli`

doom-fr, 2019-08-07:
my DNS do not know jira.arm.com

https://dnschecker.org/#A/jira.arm.com

AnotherButler, 2019-08-07:
@madchutney @mark-edgeworth Could one of y'all please help with this?

@doom-fr As a side note, are you using Mbed OS 5.11? If you're using the latest version (Mbed OS 5.13), you can find the most recent documentation here: https://os.mbed.com/docs/mbed-os/latest/tools/manual-installation.html.

doom-fr, 2019-08-07:
I have used the default `mbed-os-example-blinky`

In the new version of doc, it is still `pip install mbed-cli`, no `sudo` !

RomanSaveljev, 2019-08-07:
https://askubuntu.com/a/802594 explains why `sudo pip install` is a bad practice. I think it should not be encouraged by the official documentation. If one uses virtual environment, then plain `pip install mbed-cli` will succeed. If one does not, then calling `pip install --user` is the way to avoid `sudo`.

doom-fr, 2019-08-07:
OK but `pip install --user mbed-cli` does not work too.
(Maybe it does not work on my machine because it is not a fresh install)
Anyway, it is not written in the documentation.

And for `sudo`, Pip uses SSL. No ? So there can't be man in the middle.
Not sure of that.

But in the documentation, what's sure is that it should be written a command line that works.
And for now, it is not the case.

AnotherButler, 2019-08-09:
ping @madchutney @mark-edgeworth

mark-edgeworth, 2019-08-12:
The installation of mbed-cli is not as good as we'd like, and we're not even sure that it works on Ubuntu at the moment. We are looking at how this can be improved.  Having said that, may I suggest that you install this into a virtual python environment (eg pipenv) which may resolve some of the issues you are having.

mark-edgeworth, 2019-09-02:
Just to be clear, using sudo to force the installation is not recommended at all. It could appear to fix the problem, but it could also introduce other unexpected changes in your python environment.

AnotherButler, 2019-09-03:
Thanks for raising this issue. We believe we have solved it by adding instructions to use a virtual environment: https://github.com/ARMmbed/mbed-os-5-docs/pull/1143.

Please feel free to reopen if this issue has not been resolved.

closed 2019-09-03 by AnotherButler