ARMmbed/mbed-os-5-docs #1133: Conflicting information about Watchdog

matthew-macovsky-cypress, 2019-08-14:
The [Watchdog porting documentation](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/porting/target/Watchdog.md) states the following:
> Maximum supported timeout is UINT32_MAX milliseconds

However, the [Watchdog API documentation](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/api/drivers/Watchdog.md) states:
> The maximum amount of time you can set as the Watchdog timeout varies depending on the target hardware. You can check the maximum value by calling Watchdog::get_instance().get_max_timeout().

This seems to conflict with the porting documentation.

AnotherButler, 2019-08-14:
@jamesbeyond @0xc0170 Can one of y'all help with this?

jamesbeyond, 2019-08-15:
Hi @matthew-macovsky-cypress,

the Info here are not conflicting.

The [Watchdog porting documentation](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/porting/target/Watchdog.md) is the HAL API porting doc. this doc mean to be used by partners porting their platform to mbed-OS. In this doc we are saying that, mbed-OS allow partners given any timeout value (1ms<value<UINT32_MAX) base on their platform spec. So that means different platform will have different values.  

The [Watchdog API documentation ](https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/api/drivers/Watchdog.md) is the users API doc, mean to be used by a user when he writes his own application and use mbed watchdog API. In here the watchdog timeout value might be vary, depend on the target he choose, he need to call this function to query the exact value.

Hope that explains your question.

matthew-macovsky-cypress, 2019-08-15:
Hi @jamesbeyond, thanks for the explanation.  I think I understand better now.  Would I be correct then to interpret that line in the porting documentation as saying that `UINT32_MAX` is the maximum allowed `max_timeout` returned from `hal_watchdog_get_platform_features()`, and not as saying that vendors must support a `timeout_ms` of `UINT32_MAX` in `hal_watchdog_init()`?

jamesbeyond, 2019-08-15:
@matthew-macovsky-cypress  you are right. Vendor's target WDG max timeout usually is a fixed value determinate by the HW register bit-width and frequency etc. As long as this value within the range `1ms<value<UINT32_MAXms` , mbed-os can support this target without any issue. But if a target watchdog timeout is out of that scope. we will need some workarounds

matthew-macovsky-cypress, 2019-08-15:
Alright, thanks for your help.

closed 2019-08-15 by matthew-macovsky-cypress