ARMmbed/mbed-os-5-docs #1135: compiling-the-code.html has no windows info

mj8w, 2019-08-16:
# Docs issue template

## Page link
https://os.mbed.com/docs/mbed-os/v5.13/quick-start/compiling-the-code.html
## Expected to find...
CLI instructions for windows
## Found
Apparently Linux CLI?

mj8w, 2019-08-16:
So, I'm lost. Guess I'll find a different OS.

AnotherButler, 2019-08-16:
@mj8w How would you recommend we clarify this to minimize future confusion?

mj8w, 2019-08-16:
What happened in my case (which I should have known better), is I already had a cmd prompt window open, while it installed, and forgot that the PATH would be re set. When I tried mbed, nothing worked.
The thing is, you can't tell what the prompt (on the webpage) means, like is this the cmd prompt, or something else? Maybe I was supposed to open an mbed app and then I get this prompt?
I think the page needs to explicitly state: "When the installation is complete, open a cmd or bash shell, and enter the following..." That would have made it clear to me.

AnotherButler, 2019-08-22:
We have merged PR #1138 to resolve this issue. Please feel free to reopen it if this does not solve it.

closed 2019-08-22 by AnotherButler