ARMmbed/mbed-os-5-docs #1136: Error on NetworkInterface set_blocking() docs.

javiermelendezc, 2019-08-16:
https://os.mbed.com/docs/mbed-os/v5.13/mbed-os-api-doxy/class_network_interface.html#aaf6bf1dfffbe6a5626b7b52eaa542b6e

Documentation on connect() and disconnect() says:

> This blocks until connection is established, but asynchronous operation can be enabled by calling NetworkInterface::set_blocking(false).

Then in set_blocking() parameters description it says:

> virtual nsapi_error_t set_blocking(bool blocking)
> ....
> blocking | Use true to set NetworkInterface in asynchronous mode.

I think it should be: **Use false to set NetworkInterface in asynchronous mode.**

AnotherButler, 2019-08-19:
@SeppoTakalo Could someone on your team please review this? I think it must be fixed in the code.

AnotherButler, 2019-08-23:
ping @jarlamsa @SeppoTakalo

jarlamsa, 2019-08-26:
@AnttiKauppila has the responsibility of connectivity. And yes, it seems to be a mistake in the set_blocking doxygen.

AnttiKauppila, 2019-08-27:
https://github.com/ARMmbed/mbed-os/pull/11341