ARMmbed/mbed-os-5-docs #1137: Lack of documentation for startAdvertising

chocolatepatty, 2019-08-19:
Unable to find documentation for startAdvertising(advertising_handle_t, adv_duration_t, uint8_t).

https://os.mbed.com/docs/mbed-os/v5.13/mbed-os-api-doxy/class_gap.html#ad84f0991daf05aa3f2e1a7437d32b504

Entry for startAdvertising(void) informs that this is deprecated, but does not link to any information about function given in "use instead". Have been unable to find any useful information about this elsewhere either.

AnotherButler, 2019-08-19:
@chocolatepatty Thanks for raising this issue. I think you may find the information you're looking for here: https://os.mbed.com/docs/mbed-os/v5.13/mbed-os-api-doxy/classble_1_1_gap.html#a49774a7becc9d4fa30dfeba697ca709a

@pan- Is there any additional information on this? Also, could you please link these in the Doxygen?

pan-, 2019-08-19:
The function you're looking for is documented here: https://os.mbed.com/docs/mbed-os/v5.13/mbed-os-api-doxy/classble_1_1_gap.html#a49774a7becc9d4fa30dfeba697ca709a . We could look at why it is not linked inside the doxygen output.

chocolatepatty, 2019-08-19:
Thanks kindly for the speedy response, that looks like it should be helpful!

closed 2019-08-19 by chocolatepatty