ARMmbed/mbed-os-5-docs #1139: Specify full directory path on "Working with mbed CLI" user guide

ItsNageeK, 2019-08-20:
# Full path to MBED_OS_DIR not specific in guide
Minor clarification, but when attempting to follow the guide, at the link below, specifically for managing multiple projects, could clarify that the path to the projects directory should be the full path, not just the immediate path when setting MBED_OS_DIR global, example below

`/home/user/workspace/mbed-os`

NOT

`workspace/mbed-os`

https://os.mbed.com/docs/mbed-os/v5.13/tools/working-with-mbed-cli.html

AnotherButler, 2019-08-20:
@CuriousEngineering Thanks for raising this issue 👍 

If you're interested in contributing a fix to this or any other documentation issue, our documentation contributions follow the [same guidelines as our code](https://os.mbed.com/docs/mbed-os/v5.13/contributing/index.html) (as applicable).

AnotherButler, 2019-08-22:
We have merged PR #1141 to resolve this issue.

closed 2019-08-22 by AnotherButler