ARMmbed/mbed-os-5-docs #1149: debugging in vscode with openocd

mlajus, 2019-09-22:
The [mbed+vscode tutorial](https://os.mbed.com/docs/mbed-os/v5.13/tutorials/visual-studio-code.html) is a bit laconic about openocd debugging configuration.

As I'm not an expert, I had  to dig and search quite a bit to do it work with a nucleo stm32f4 board :

Here are the important settings I used in launch.json :
"debugServerArgs": "-f board/st_nucleo_f4.cfg -c init -c \"reset init\"",
"serverStarted": "Info : stm32f",
"MIDebuggerPath": "/usr/local/bin/arm-none-eabi-gdb",
"debugServerPath": "openocd"

The second line is not obvious per se. It is required by the launch script to know when the debugger has confirmed the link with the target board, in order to continue the debug process.
What I understood is that it has to match a character string in the debugger stdout when the debug link with the target is correctly established.
It should be adapted depending on the target used.

It would be better it that was explained in the tutorial.

AnotherButler, 2019-09-30:
@mlajus Thanks for raising this issue. @evedon Is this something your team could look into?

evedon, 2019-09-30:
This should be covered by OTG team @thegecko