ARMmbed/mbed-os-5-docs #1169: Mbed OS Docs are out of date, confusing and misleading.

star297, 2019-11-12:
# Docs issue template
Mbed-os v5.14 docs and examples appear to be from earlier versions.
It's less confusing to have a blank page indicating an 'under construction' or 'Documentation not ready' label on these pages.
The example indicates the use of 'wait' however the api information states depreciated since v5.14    

## Page link
https://os.mbed.com/docs/mbed-os/v5.14/apis/wait.html
https://os.mbed.com/docs/mbed-os/v5.14/mbed-os-api-doxy/mbed__wait__api_8h_source.html

## Expected to find...
Correct information

## Found
Out of date and depreciated information

AnotherButler, 2019-11-13:
Thank you for raising this issue. We're looking into it now.

AnotherButler, 2019-12-05:
We've created PRs #1170 and #1180 to address this issue. Please feel free to reopen it or open a new one if you have further questions.

closed 2019-12-05 by AnotherButler

hadefuwa, 2021-05-29:
I'm so put off by Mbed due to the sheer amount of outdated and deprecated articles.

0xc0170, 2021-05-31:
> I'm so put off by Mbed due to the sheer amount of outdated and deprecated articles.

Can you provide examples , what articles shall we clean up? You can create a new issue with more details that could help us to fix them.