ARMmbed/mbed-os-5-docs #1179: Mbed OS build rules

tdjastrzebski, 2019-12-03:
In [Mbed OS build rules](https://os.mbed.com/docs/mbed-os/v5.10/tools/mbed-os-build-rules.html) I found nothing about **include** folders.
**mbed-os** contains 10+ **include** folders. It seems those folders contain headers which are not referenced (neither directly nor indirectly) but which have to be included.
This is undocumented.

AnotherButler, 2020-01-02:
@madchutney @mark-edgeworth Could one of you please point me to or provide me with content about include folders, so I can add it to the documentation?

0xc0170, 2020-01-03:
The link referenced above contain sources, but might be worth documenting what header extensions are supported.

Everything that is not skipped is included - every header file found in these folders is being added (following rules about TARGET, FEATURE, COMPONENT, TOOLCHAIN). A simple note might added to the build rules?

tdjastrzebski, 2020-01-04:
@0xc0170 @AnotherButler I am not certain it is worth starting a new thread but here is another build issue.
`mbed-os\features\cryptocell\FEATURE_CRYPTOCELL310\include\ccsw_crys_rsa_shared_types.h` header uses `uint32_t` type while this header file has no `#include` directives at all - `stdint.h` would be needed.
That makes me believe thare are some additional undocumented build rules which must be followed. In this case probably `stdint.h` should be forcefully included but I am not certain. Maybe it is just a bug.

mark-edgeworth, 2020-01-06:
Regarding this last point. I don't think the build system is doing anything strange here. Whenever ccsw_crys_rsa_shared_types.h is included in a source unit, there will be a preceding #include of something like stdint.h that defines uint32_t. The header is not compiled on its own.

tdjastrzebski, 2020-01-07:
@mark-edgeworth I suspect that the order in which header files are forcefully included matters. Otherwise `ccsw_crys_rsa_shared_types.h` would not compile. If this is the case I strongly suggest following [headers best practices](https://www.acodersjourney.com/top-10-c-header-file-mistakes-and-how-to-fix-them/) - see **MISTAKE # 4: Not making the header file compliable by itself**

tdjastrzebski, 2020-01-07:
@AnotherButler Probably there are some other undocumented build rules. E.g. `objects_cryptocell.h` header is not included anywhere, does not lay in `includes` folder but definitely is needed by `trng.c`.

mark-edgeworth, 2020-01-13:
> @mark-edgeworth I suspect that the order in which header files are forcefully included matters. Otherwise `ccsw_crys_rsa_shared_types.h` would not compile. If this is the case, I suggest following [headers best practices](https://www.acodersjourney.com/top-10-c-header-file-mistakes-and-how-to-fix-them/) - see **MISTAKE # 4: Not making the header file compliable by itself**.
> New issue #1196 created

Completely agree, although this is outside of the build tools remit I'm afraid.

tdjastrzebski, 2020-01-14:
I reported another potential code issue: ARMmbed/mbed-os#12252

AnotherButler, 2020-03-02:
I've put up this PR to address the initial concern: https://github.com/ARMmbed/mbed-os-5-docs/pull/1260

mark-edgeworth, 2020-03-03:
As hinted at earlier, the correct fix for this is to correct the header file(s) in question so that they include whatever other headers that they require to provide all of the type definitions etc needed.

AnotherButler, 2020-03-03:
Thanks for clarifying. I thought when @0xc0170 said "A simple note might added to the build rules?", that was the best fix. 

Because the only fix is changes to the header files, and it looks like https://github.com/ARMmbed/mbed-os/issues/12244 addresses this, shall I close this issue?

mark-edgeworth, 2020-03-05:
Yes, I reckon so. Thanks A!

closed 2020-03-05 by AnotherButler

tdjastrzebski, 2020-03-14:
@AnotherButler by tracing what exactly `mbed` Python build script is doing I figured out that actually all paths have to be included.