ARMmbed/mbed-os-5-docs #1186: Thread docs need to deprecate Wait ?

LMESTM, 2019-12-11:
# Docs issue template

Hasn't wait API been deprecated and replaced by sleep_for in latest versions ? 

## Page link
https://os.mbed.com/docs/mbed-os/v5.9/reference/thread.html

## Expected to find...
update on deprecated and recommended APIs

## Found

AnotherButler, 2019-12-11:
@LMESTM Thanks for raising this issue. That page links to the 5.9 documentation. We have added updates in the latest documentation, which will go out in 5.15.

LMESTM, 2019-12-11:
@AnotherButler 
Do you mean there should be another up-to-date page like : 
https://os.mbed.com/docs/mbed-os/v5.15/reference/thread.html
I haven't found any more recent version than the V5.9  ...

kjbracey-arm, 2019-12-11:
Latest version is here: https://os.mbed.com/docs/mbed-os/v5.14/apis/thread.html

AnotherButler, 2019-12-18:
Please feel free to reopen this issue if the link provided in the most recent comment doesn't resolve this issue.

closed 2019-12-18 by AnotherButler