ARMmbed/mbed-os-5-docs #1190: period of event1 is reset by cancel

noritan, 2020-01-01:
# Docs issue template
The example program does not work as expected.

## Page link
https://os.mbed.com/docs/mbed-os/v5.15/apis/eventqueue.html#static-eventqueue-example-posting-user-allocated-events-to-the-static-queue

## Expected to find...
It is described in the example code that two set of events are dispatched.  But the event1 is dispatched only once.

## Found
The period parameter of event1 is set to 200 in the main() function.  But the parameter is reset when the event1 is cancelled by queue.cancel(&event1);  An additional event1.period(200); must be called after the cancel.

maciejbocianski, 2020-01-22:
Example code is Ok.
I found bug in UserAllocatedEvent implementation.
Working on it...

AnotherButler, 2020-03-20:
@maciejbocianski What's the status of this? Has it been resolved?

maciejbocianski, 2020-03-20:
@AnotherButler  It has been fixed see https://github.com/ARMmbed/mbed-os/pull/12319

AnotherButler, 2020-03-20:
@maciejbocianski Thanks for confirming 👍 

@noritan Does this resolve the issue for you?

noritan, 2020-03-21:
The code change looks good.
But, I didn't try to run the example on 6.0.0-alpha

AnotherButler, 2020-03-26:
We believe we have resolved this, so I'm closing this issue. Please reopen if it is not resolved.

closed 2020-03-26 by AnotherButler