ARMmbed/mbed-os-5-docs #1191: Example program causes run-time error to stop.

noritan, 2020-01-01:
# Docs issue template

## Page link
https://os.mbed.com/docs/mbed-os/v5.15/apis/eventqueue.html#eventqueue-example-deferring-from-interrupt-context

## Expected to find...
Expected to execute the example program without any problem.

## Found
When this example started and the push button is pushed, the example stops with following message and blinking LED1 like "----...." .

```
Starting in context 080047ac
fall_handler in context 080051bc

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x8003C6C, Not allowed in ISR context
Location: 0x100127C1
Error Value: 0x8003C6C
Current Thread: rtx_idle Id: 0x80047F0 Entry: 0x10010355 StackSize: 0x300 StackMem: 0x8004878 SP: 0x80476F4
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=CY8CKIT_062_WIFI_BT
-- MbedOS Error Info --
```

maciejbocianski, 2020-01-17:
Working on it...
Fixed test will be moved to [mbed-os-examples-docs_only](https://github.com/ARMmbed/mbed-os-examples-docs_only)
https://github.com/ARMmbed/mbed-os-examples-docs_only/pull/63

maciejbocianski, 2020-01-22:
Here is fixed version https://github.com/ARMmbed/mbed-os-examples-docs_only/pull/63

maciejbocianski, 2020-01-24:
Doc link updated https://github.com/ARMmbed/mbed-os-5-docs/pull/1198

AnotherButler, 2020-02-13:
PR #1198 has merged, so I'm closing this issue. Please reopen it if it needs further resolution.

closed 2020-02-13 by AnotherButler