ARMmbed/mbed-os-5-docs #1192: The size argument for EventQueue is too small.

noritan, 2020-01-01:
# Docs issue template
No events are created due to less size of EventQueue storage.

## Page link
https://os.mbed.com/docs/mbed-os/v5.15/tutorials/the-eventqueue-api.html#calculating-the-number-of-events

## Expected to find...
The example program can be executed without any issues.
Three calls succeed and last one fails. 

## Found
In the example program an EventQueue instance is created with an argument size=8*sizeof(int)  This allocates 8 word of storage but it is not enough to to create an Event object.  At least size=2*EVENTS_EVENT_SIZE is required in this example.

In addition, no failures can be detected by this example program because the call() method only returns ZERO when failed to allocate a memory and never shows error messages.  It is required to show the return value of the call() method like follows.

```
printf("call(1)=%d\r\n", queue.call(func, 1));       // requires 2 words of storage
```

jamesbeyond, 2020-01-10:
Thanks for reporting the issue, we are looking into this ASAP

maciejbocianski, 2020-01-24:
fix is here https://github.com/ARMmbed/mbed-os-5-docs/pull/1199

AnotherButler, 2020-02-04:
Thanks again for raising this issue. This PR has merged, so we are closing this issue. Please feel free to reopen it if it has not been resolved.

closed 2020-02-04 by AnotherButler