ARMmbed/mbed-os-5-docs #1194: CLI OS X Download link broken.

KiwiHC16, 2020-01-02:
# Docs issue template
Can't get CLI for OS X

## Page link
https://os.mbed.com/docs/mbed-os/v5.9/tools/installing-with-the-mac-os-x-installer.html
Download the Mac OS X installer for Mbed CLI.

## Expected to find...
CLI dmg

## Found
XML file:
```
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>NoSuchKey</Code>
<Message>The specified key does not exist.</Message>
<Key>builds/mbed-cli-v0.0.10.dmg</Key>
<RequestId>444434488B38D9A9</RequestId>
<HostId>
TJjb1PQaGfBO3n5wb6m5PBcJN0EOJ18keHpm0JAu6exulkkdW/RtCQPHJC2vCN/CYJkUBMeDvfQ=
</HostId>
</Error>
```

AnotherButler, 2020-01-02:
@KiwiHC16 Thank you for raising this issue. This link is from Mbed OS 5.9 and is no longer up to date. We recommend using the latest installer release, which you can find at https://github.com/ARMmbed/mbed-cli-osx-installer/releases/latest.

If you are using a more recent version of Mbed OS, you can find our latest Mbed CLI documentation at https://os.mbed.com/docs/mbed-os/v5.15/tools/installation-and-setup.html

KiwiHC16, 2020-01-06:
Thanks a lot.

AnotherButler, 2020-01-08:
We believe we have resolved this issue. Please feel free to reopen it has not been resolved.

closed 2020-01-08 by AnotherButler