ARMmbed/mbed-os-5-docs #1196: Make header files compliable by themselves

tdjastrzebski, 2020-01-07:
I suspect that the order in which header files are forcefully included during build matters. Otherwise `ccsw_crys_rsa_shared_types.h` would not compile since it uses `uint32_t` type without including `<sdtint.h>`.
If this is the case, I suggest following [headers best practices](https://www.acodersjourney.com/top-10-c-header-file-mistakes-and-how-to-fix-them/) - MISTAKE # 4: Not making the header file compliable by itself.

AnotherButler, 2020-01-08:
@tdjastrzebski  This looks like a code issue, not a documentation issue. Could you please raise this issue against the [`mbed-os` repository](https://github.com/ARMmbed/mbed-os/issues)?

mark-edgeworth, 2020-01-13:
Yes, this is a code issue.

tdjastrzebski, 2020-01-13:
Issue #12244 has been reported at https://github.com/ARMmbed/mbed-os/issues

AnotherButler, 2020-01-13:
@tdjastrzebski Thank you. I'll close this issue here, so we can follow it in the new `mbed-os` issue you've raised.

closed 2020-01-13 by AnotherButler