ARMmbed/mbed-os-5-docs #1217: PSA Attestation API page doesn't specify the spec version

adrianlshaw, 2020-02-03:
## Page link

https://os.mbed.com/docs/mbed-os/v5.15/apis/psa-initial-attestation.html

## Expected to find...

The version of this PSA API implemented is missing. The API version on this page is 1.0 beta0.

This is important for end users to note because there was a breaking change between 1.0 beta0 and 1.0 stable. TF-M is also in the processing of merging 1.0 stable into upstream, so it would be best to avoid confusion (especially since this is the second result on Google search). 

Also there ought to be a reference to the spec document identifier (ARM IHI 0085).

## Found

Nothing.

AnotherButler, 2020-02-05:
@adrianlshaw Could you please elaborate more? Which link is out of date? The link to the TF-M repository? Or is the Doxygen, itself, out of date?

adrianlshaw, 2020-02-05:
No links are out of date. I am just saying that the text at the top of the page should specify the version of the API.

So replace:

"The initial attestation service exposes the following PSA interfaces"
 
with

"The initial attestation service exposes the following interfaces (based on the Beta-0 spec of the PSA Attestation API)"

AnotherButler, 2020-02-05:
Thanks for clarifying 👍 

When will we need to do something with `1.0 stable`?

adrianlshaw, 2020-02-05:
> When will we need to do something with 1.0 stable?

As soon as Mbed implements it :-) but I don't know when that might happen (I'm not in ISG).

AnotherButler, 2020-02-13:
PR #1221 has merged, so I'm closing this issue. Please reopen it if it needs further resolution.

closed 2020-02-13 by AnotherButler