ARMmbed/mbed-os-5-docs #1233: Target repository referenced does not exist

keepitsimplejim, 2020-02-15:
Repository referenced to for the baseline target template sends you to a repository that does not exits
## Page link
https://os.mbed.com/docs/mbed-os/v5.15/porting/porting.html
## Expected to find...
Template for a new target at https://github.com/ARMmbed/mbed-os-new-target 
## Found
404 page not found.

AnotherButler, 2020-02-25:
@keepitsimplejim Thank you for raising this issue. Yes, this shows how to name the branch, but I see how that's not clear. I'll put up a PR clarifying this.

AnotherButler, 2020-03-02:
I've put up PR 1261 to resolve this issue. Please take a look, and let us know if this makes more sense.

AnotherButler, 2020-03-16:
We believe we have fixed this issue. Please feel free to reopen this ticket or start a new one if further changes are needed.

closed 2020-03-16 by AnotherButler