ARMmbed/mbed-os-5-docs #1234: Return value of Mail#get() method is incorrect

noritan, 2020-02-16:
# Docs issue template

## Page link
https://os.mbed.com/docs/mbed-os/v5.15/mbed-os-api-doxy/classrtos_1_1_mail.html#ad033ffca23a9800b6607f08cec807417

## Expected to find...
Mail#get() returns an osEvent structure with osEventMail in the status field.

## Found
The reference manual said osEventMessage is returned as the return value, but this is incorrect.  The return value is an osEvent structure with status osEventMail as described in the Mail example https://os.mbed.com/docs/mbed-os/v5.15/apis/mail.html#mail-example

AnotherButler, 2020-02-18:
@noritan Thank you for raising this issue 👍 Could you please explain further? Specifically, is it the reference API page that needs to be updated (https://os.mbed.com/docs/mbed-os/v5.15/apis/mail.html) or the Doxygen page (https://os.mbed.com/docs/mbed-os/v5.15/mbed-os-api-doxy/classrtos_1_1_mail.html#ad033ffca23a9800b6607f08cec807417)?

noritan, 2020-02-19:
There are two points to be corrected in the **Return values** description.

1. The method returns an osEvent structure instead of an integer value.  The integer status code is stored in the status field.
2. The status field is osEventMail when message received.

The Doxgen page says in the declaration line that the Mail#get() method returns osEvent type structure.

`osEvent get (uint32_t millisec = osWaitForever)`

But the **Return values** description says it can return `osEventMessage` which is an integer value.  All other possible **Return values** are integer.

Status | Description
-- | --
osEventMessage | Message received.
osOK | No mail is available (and no timeout was specified).
osEventTimeout | No mail has arrived during the given timeout period.
osErrorParameter | A parameter is invalid or outside of a permitted range.

The actual implementation is "Mail#get() returns an osEvent structure which status field is osEventMail when a message received."

The description regarding the return value is described in the Doxgen page which is linked from the reference API page.  I don't know how is the documents generated, but I found same description in the source code Mail.cpp comment  So, the source code should be corrected first.

AnotherButler, 2020-02-25:
@evedon Could someone on your team please help with this?

evedon, 2020-02-27:
@AnotherButler I will move the JIRA ticket to the team's backlog.

closed 2020-03-16 by 0xc0170