ARMmbed/mbed-os-5-docs #1281: List of supported target controllers missing.

dhebbeker, 2020-03-18:
## Expected to find...

List of all supported micro-controllers (for Mbed OS bare metal profile). For example listing the STM32F401.

## Found

* [List of boards](https://os.mbed.com/platforms/)
* [List of modules](https://os.mbed.com/modules/)
* [List of components](https://os.mbed.com/components/)

AnotherButler, 2020-03-19:
@MarceloSalazar Can your team help with this?

MarceloSalazar, 2020-03-20:
We expect to publish the list of baremetal boards in os.mbed.com/platforms for Mbed 6+

AnotherButler, 2020-03-20:
Thanks @MarceloSalazar

dhebbeker, 2020-03-20:
> We expect to publish the list of baremetal boards in os.mbed.com/platforms for Mbed 6+

Thank you @MarceloSalazar. I will leave this issue to be closed with the publication of the list, if this is ok?

iriark01, 2020-06-04:
Mbed OS 6 filter is now available: https://os.mbed.com/platforms/?q=&Mbed+OS+support=Mbed+OS+6.0
Does that resolve this issue, or will there be a specific bare metal filter?

MarceloSalazar, 2020-06-04:
We'll be adding a flag for both RTOS and Bare metal shortly (it may take a bit).
At the moment the most reliable source of information is [Github - targets.json](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json) - look for "bare-metal"

dhebbeker, 2020-06-04:
> Does that resolve this issue, or will there be a specific bare metal filter?

I checked again and still did not find a list of the supported *micro-controllers* (not boards). Thus I would not consider this issues as resolved yet.

> At the moment the most reliable source of information is [Github - targets.json](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json) - look for "bare-metal"

This is indeed the closest I could find to a list. It would be nice to have it as a proper list in the documentation.

lazyoracle, 2021-05-09:
There now seems to be filters for both RTOS and bare-metal in the list of boards on the mbed website [here](https://os.mbed.com/platforms/?q=&Mbed+OS+6=Bare+metal&Target+vendor=STMicroelectronics). I believe this can now be considered resolved and ready to be closed?

dhebbeker, 2021-05-18:
I opened the issue as I was expecting a list of all microcontrollers which are supported by Mbed OS. 

Please consider this use case: I already have a proprietary board design using some controller X. Now I would like to look up, if this controller is supported by Mbed OS. As far as I know, the best option at the moment is to search for that controller in the list, as @MarceloSalazar [suggested](https://github.com/ARMmbed/mbed-os-5-docs/issues/1281#issuecomment-638760501):

> At the moment the most reliable source of information is [Github - targets.json](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json) - look for "bare-metal"

I am still missing, a comprehensible (dedicated) list of supported microcontrollers - *not boards*.