ARMmbed/mbed-os-5-docs #1291: The EventQueue API Tutorial

mertkapar, 2020-04-08:
Hi,
I want to ask something related to EventQueue tutorial with the header **Calculating the number of events** 

https://os.mbed.com/docs/mbed-os/v5.15/tutorials/the-eventqueue-api.html

I am assuming that 'func' has same size constraints with 'func3' and after the `queue.dispatch();`
all memory is freed and calling 

1.  `queue.call(func, 1, 2, 3); // requires 13 words of storage (9+4)`
2. `queue.call(func, 1, 2, 3); //  fails` 

So total word size 26 and `queue.dispatch();` has called. 
I think allocation **succeeded**

 

I think the given example is wrong or am I missing something ?

1. `void func1(int);`
2. `void func3(int, int, int);`
3. `EventQueue queue(2*(9+4)*sizeof(int)); // 26 words of storage (store two callbacks with three arguments at max)`
4. `queue.call(func1, 1);       // requires 11 words of storage (9+2)`
5. `queue.call(func3, 1, 2, 3); // requires 13 words of storage (9+4)`
6. `// after this we have 2 words of storage left`
7. `queue.dispatch(); // free all pending events` 

After the `queue.dispatch();` all memory is free but when you look at the comments 

8. `queue.call(func, 1, 2, 3); // requires 13 words of storage (9+4)`
9. `queue.call(func, 1, 2, 3); // fails` 

and other comments 

10. `// storage has been fragmented into two events with 11 and 13 words`
11. `// of storage, no space is left for an another 13 word event even though two words`
12. `// exist in the memory region`

evedon, 2020-04-23:
This tutorial gives an example of memory fragmentation when allocating events. 

There are two main things to keep in mind:
- successfully allocated memory is never fragmented further
- untouched space can service any event that fits

So in the example above, a `queue` is created with space for 26 words. Two events are allocated which take 11 and 13 words of storage respectively. At this point, storage has been fragmented into 3 blocks of 11 (event 1), 13 (event 2) and 2 (free space) words. 

At line 7, the two events are freed so the event queue storage has space for 26 words in total but fragmented into  3 blocks of 11, 13 and 2 words.
At line 8, an event is allocated to store the 13 words required.
At line 9, the call fails because there are no free block of size 13 left.

I will close this as I believe I have answered your question, please let me know if I can clarify this further.

closed 2020-04-23 by evedon

iriark01, 2020-06-02:
@evedon should we update the docs?

evedon, 2020-06-02:
The information was already in the tutorial so I don't think we need an update.