ARMmbed/mbed-os-5-docs #1332: Remove/update references to the Windows serial driver

MarceloSalazar, 2020-06-01:
# Docs issue template

The Serial driver is no longer needed on Windows 10.
In fact, it creates a problem when trying to use WebUSB.
See https://github.com/ARMmbed/DAPLink/issues/487
I also hit this problem myself and it took me a while to find out.

## Page link

https://os.mbed.com/docs/mbed-os/v5.15/tutorials/windows-serial-driver.html
And possibly many more that refer to windows serial driver.

## Expected to find...

Details that the driver is needed for old versions of Windows (7 and older) and it's not required for Windows 10.
Discourage the installation of the serial driver.

## Found

Recommendations to install the driver on Windows

---

FYI @iriark01 @thegecko @arekzaluski

thegecko, 2020-06-02:
It's also mentioned on the DAPLink page of mbed.com (albeit outlining not Windows 10 in the text):

https://os.mbed.com/handbook/DAPLink

As Windows 7 is EOL and no longer supported and this shouldn't be used on newer Windows, what would be the impact of removing it altogether?

At the very least, we need a large warning banner in both places for users to not use it on Windows 8.1 or 10.

@JoeTheGuitarist thoughts?

JoeTheGuitarist, 2020-06-02:
It should ideally be removed altogether, but starting with a clear warning message anywhere the drivers are mentioned would suffice for now. FYI - Windows 7 represents only about 10% of all Windows usage of our website and has declined significantly in the last 6 months.

LDong-Arm, 2020-06-02:
@MarceloSalazar Thanks for raising this issue.

@thegecko @JoeTheGuitarist @arekzaluski 
I guess we can remove it from the documentation altogether. The [Mbed CLI Windows installer](https://github.com/ARMmbed/mbed-cli-windows-installer/releases) already includes the serial driver and gets installed (or not) depending on the Windows version. The user shouldn't need to do anything about it at all.

I'll update the documentation.

iriark01, 2020-06-02:
What if they're not using Mbed CLI? Will they have a problem if they're using the Online Compiler?

MarceloSalazar, 2020-06-02:
Note the Mbed CLI installed hasn't been updated for a while and may never do. 
Instead of relying on the installer, I'd suggest we informed that the driver is only required for Windows 7 and older versions.

LDong-Arm, 2020-06-02:
Okay, fair point

LDong-Arm, 2020-06-02:
I've created #1334.

As for https://os.mbed.com/handbook/DAPLink, it's an old Mbed OS 2 page which may not be updated anymore?

PR #1334 (2020-06-03): A serial driver should be installed on Windows 7 only
Fixes #1332 

This PR
* stresses on the serial driver doc page that it only applies to Windows 7
* update other docs that link to the serial driver doc page
closed 2020-06-03 by iriark01