ARMmbed/mbed-os-5-docs #1340: Duplicated content about custom boards

MarceloSalazar, 2020-06-05:
## Page link

https://os.mbed.com/docs/mbed-os/v6.0/program-setup/adding-and-configuring-targets.html
https://os.mbed.com/docs/mbed-os/v6.0/porting/derivative-port.html

## Expected to find...

A single guide on using custom board support.

## Found

Duplicate information.

It would be great to unify the content and deliver it in the porting area.
I think it is a bit too low level / advanced to be included in the 'program setup' area but a link to redirect would be sufficient.

donatieng, 2020-06-09:
Totally agree - generally this section needs major improvements, let's make sure this gets captured as part of https://jira.arm.com/browse/IOTOSM-11?jql=text%20~%20%22custom%20targets%22%20and%20project%20%3D%20IOT-OS-M

iriark01, 2020-06-12:
I have a question. 

Custom targets in Mbed Studio are not part of porting in any way - they're intended for debug probes and other "transient" modifications. Why are custom targets in Mbed OS part of porting?

https://os.mbed.com/docs/mbed-studio/current/mbed-os/custom-targets.html

@donatieng @MarceloSalazar

MarceloSalazar, 2020-06-15:
hi @iriark01 
I think there are different approaches to custom boards. I'll share my view and I'm open to discussing further.

In Mbed Studio, the concept of custom boards applies to: 
- Boards with Mbed Enabled compliant interfaces (e.g. STLink or DAPLink) - either standard boards listed in our website or custom designs that re-use the same interface (for instance you can 'break' a NUCLEO board and wire the STLink to a custom board).
- Boards with standard CMSIS-DAP probes (e.g. [STLink/V2](https://www.st.com/en/development-tools/st-link-v2.html) adapter) that may not be Mbed Enabled compliant but are very common and useful.
Additionally, Mbed Studio (and Mbed CLI) just build the custom board configuration without requiring the user to do anything else (apart for creating the configuration, as explained next).

Now on the Mbed OS side, "creating" a custom board requires dealing with more advanced concepts related to MCU/board specific features, pin-outs, possibly custom memory maps and others, **before** it can be used to build an actual application. Many items related to porting, configuration, and testing need to be considered. This is why I'm inclined to keep custom boards in the porting area.