ARMmbed/mbed-os-5-docs #1345: Invalid reference to pelion E2E tests

maclobdell, 2020-06-29:
## Page link

https://www.pelion.com/docs/device-management/current/porting/test-pal.html

## Expected to find...

An explanation of E2E tests or a link to a valid page.

## Found
"Detailed information on the E2E tests and their execution is available in the E2E tests README document."

README document link is bad. It goes to this, which is invalid. 
https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/TESTS/E2E/README.md

## Other details
There is useful information in these documents. Perhaps the first one is the right one.
https://github.com/ARMmbed/mbed-cloud-client-example/blob/master/TESTS/README.md
https://github.com/ARMmbed/pelion-e2e-python-test-library/blob/master/README.md

However, I think that it would be good to include some of the key information in the documentation and not just rely on readme files.

iriark01, 2020-06-30:
@AnotherButler this is for Pelion docs, not Mbed OS. Please review and close this issue when you've moved it across.

AnotherButler, 2020-06-30:
As a first step, I have fixed the link on versions 2.2 through 4.5, so it no longer 404s.

iriark01, 2020-07-03:
Thanks, Amanda. 

I am going to close this, as it's on the wrong GitHub repository.

closed 2020-07-03 by iriark01

maclobdell, 2020-07-08:
oops, my bad. wrong repo. thanks