ARMmbed/mbed-os-5-docs #1353: Event API documentation outdated

boraozgen, 2020-07-28:
# Docs issue template

## Page link
https://os.mbed.com/docs/mbed-os/v6.2/apis/event-handling-apis.html

## Expected to find...
Better explanation of Event API, e.g. how to use the chrono durations, how to disable the period of an event. 

## Found
Examples using the deprecated API, i.e. integer milliseconds instead of the chrono arguments.
Furthermore, class reference is not visible in the page.

adbridge, 2020-11-26:
Hi @boraozgen I'm looking at updating our documentation for Events. I just wanted to clarify what you mean by:
" how to disable the period of an event". The primary functions currently available for events are:
delay()
period()
cancel()

The cancel() function seems to be closest to what you are asking but it only attempts to cancel the last event posted to the queue.... Are you looking for something else ?

boraozgen, 2020-11-26:
Hi,

I meant how to set the event to one-time after creating a periodic event. Apparently `duration(-1)` must be given to `period(),` but this is not documented.

FWeissenb, 2023-01-12:
It's not only Event API that is outdated. Threads and MemoryPool also use deprecated API. So do we need to fill another issue for this or is it okay to note this here and rename issue to "API documentation outdated"?

FWeissenb, 2023-01-12:
ok.. didn't saw the issue is already 2 years old. so @adbridge .. whats the current status? :)

adbridge, 2023-01-13:
@FWeissenb sorry I no longer work on Mbed so unfortunately cannot answer your question :)

FWeissenb, 2023-01-13:
@adbridge  any suggestions who could highlighted to answer this question? :)