ARMmbed/mbed-os-5-docs #1355: Bad link to tutorial about optimizing binary size

loverdeg, 2020-08-05:
# Docs issue template

## Page link

https://os.mbed.com/docs/mbed-os/v6.2/quick-start/further-reading.html

## Expected to find...

A tutorial on optimizing binary size.

## Found

[image]

The link went to: https://os.mbed.com/docs/mbed-os/v6.2/apis/platform-tutorials.html instead

iriark01, 2020-08-06:
Thank you for your feedback. We do not have a tutorial explicitly dealing with binary size; the link text is wrong, and I will correct it to say "Optimizing memory and power".

iriark01, 2020-08-06:
Link text fixed: https://os.mbed.com/docs/mbed-os/v6.2/quick-start/further-reading.html

closed 2020-08-06 by iriark01