ARMmbed/mbed-os-5-docs #1400: Please update stale hyperlink for Googletest primer

noonfom, 2021-01-27:
The Googletest primer has been moved to the top-level [docs](https://github.com/google/googletest/blob/master/docs) directory, so the hyperlink on this page is stale. 

## Page link
https://os.mbed.com/docs/mbed-os/v6.6/debug-test/unit-testing.html

## Expected to find...
[documentation for Google Test](https://github.com/google/googletest/blob/master/docs/primer.md)

## Found
[documentation for Google Test](https://github.com/google/googletest/blob/master/googletest/docs/primer.md)

0xc0170, 2021-01-27:
@noonfom Can you send a pull request fixing this url?

noonfom, 2021-01-27:
@0xc0170 Sure. I opened #1402 to resolve this issue.

ChrisGooch-Arm, 2022-03-24:
The above fix was accepted and merged and published as of v6.8:

https://os.mbed.com/docs/mbed-os/v6.8/debug-test/unit-testing.html

closed 2022-03-24 by ChrisGooch-Arm