ARMmbed/mbed-os-5-docs #1413: mbed-tools new . doesn't work because set(APP_TARGET ) in CMakeLists.txt

adepier, 2021-02-10:
# Docs issue template
when it says : "If you have already created a project folder, you can use ."  
## Page link
https://os.mbed.com/docs/mbed-os/v6.6/build-tools/use.html
## Expected to find...
you should add a precision "then complete in CMakeLists.txt, the line set(APP_TARGET "project folder")
## Found

0xc0170, 2021-02-10:
@adepier Hello, what is the issue?

APP_TARGET should not be required in the app if you are using new mbed-tools and latest Mbed OS master. There is a bug we are fixing (it should be fixed tomorrow on master).

0xc0170, 2021-02-10:
As this is not a bug in docs, I'll close this

This is the fix : https://github.com/ARMmbed/mbed-os/pull/14265

closed 2021-02-10 by 0xc0170