ARMmbed/mbed-os-5-docs #1445: Define RTX

BurtHarris, 2021-04-18:
# Docs issue template

## Page link
https://os.mbed.com/docs/mbed-os/v6.9/introduction/index.html contains an introduction to using the terms RTOS and RTX without introduction or definition.   The glossary at https://os.mbed.com/docs/mbed-os/v6.9/introduction/glossary.html contains an entry for RTOS, but ...

## Expected to find...
RTX seems undefined.    

## Found
Seems like platform-specific jargon, and is hard to understand for developers familiar with another platform.

BurtHarris, 2021-04-18:
@ciarmcom To clarify your reference to an _up to date clone of the repository_, please recognize I've not cloned this repository at all.    

I was reading docs online from https://os.mbed.com/docs, followed the [Report an issue on github](https://github.com/ARMmbed/mbed-os-5-docs/issues)" and pressed the [New issue ](https://github.com/ARMmbed/mbed-os-5-docs/issues/new) button.   If the formatting of the issue doesn't meet your expectations, perhaps you should add instructions to the issue template.

0xc0170, 2021-04-19:
Hi, @ciarmcom is auto bot checking the issue template and mirroring issues to internal tracker.

RTX should be added to glossary. Taken from the official RTX page:

> Keil RTX5 is a royalty-free, deterministic, full-featured real-time operating system implementing the CMSIS-RTOS API v2

ChrisGooch-Arm, 2022-03-24:
This was fixed in 6.13 onward:

https://os.mbed.com/docs/mbed-os/v6.13/introduction/glossary.html